project-generator/project_generator #436: Docs - state tools versions (the templates, what we test with)

0xc0170, 2016-08-10:
We should state in the docs, what versions are used for the templates, also for tools. What versions we use for testing. For instance for IAR it would be 7.40.2 (I used that for the template).

cc @theotherjimmy @sarahmarshy

sarahmarshy, 2016-08-10:
I agree, however, IAR only keeps downloads for its latest versions, so it will be difficult for me (and users) to find 7.40.2. I have 7.40.3, which I picked up from Rus. 

We should consider updating our IAR support to the latest version.

0xc0170, 2016-08-10:
Not updating often (every single release) but we should strive for the latest versions (to get latest MCU, and breaking changes to be resolved). I don't know yet what's their release cycle for tools, but IAR releases patches quite often.

You can download earlier versions if you got a license (need to log in to their portal), where you can find all info about versions available for your license.