project-generator/project_generator #438: Cannot enable Debug on generated Makefile

MarceloSalazar, 2016-08-15:
Tools version:
mbed CLI = 0.9.1
project_generator = 0.9.9

I expect to be able to enable DEBUG on the generated Makefile by defining `DEBUG = 1` before this `ifeq`:

```
ifeq ($(DEBUG), 1)
  CC_FLAGS += -DDEBUG -O0
else
  CC_FLAGS += -DNDEBUG -Os
endif
```

However, debug symbols are not enabled because:
1. There is no option to add `-DDEBUG -O0` to `CPPC_FLAGS` in the `ifeq` statement (e.g. `main.cpp`)
2. `CC_FLAGS` and `CPPC_FLAGS` are already defined earlier with `-Os` option:
```
CC_FLAGS = -c ... -Os ...
CPPC_FLAGS = -c ... -Os ...
```

theotherjimmy, 2016-08-15:
Actually, the flag to add debug information to GCC's output files is `-g` it is simply missing from the CC_FLAGS variable.

0xc0170, 2016-08-15:
Note: we should port makefiles in mbed. Because mbed does not currently use progen, thus this issue is mbed related, not progen.

theotherjimmy, 2016-08-15:
@0xc0170 is right, this should be moved to armmbed/mbed-os .

closed 2016-08-16 by 0xc0170