project-generator/project_generator #445: Progen for Daplink with Python 3 support is not compiling source files

brianesquilona, 2018-09-17:
uvision project files generated by progen is different between Python 2 and 3. Also in python 3, the ide does not show file names.

To replicate:

1. Use PR https://github.com/ARMmbed/DAPLink/pull/499 for Daplink python 3 support.
2. Create a virtualenv, for both Python 2 and 3, i.e. env27 and env37
3. Activate each virtualenv and run tools\launch_uvision.bat
- tools\launch_uvision.bat env27 requirements.txt
Check the project files from ide or do a compile. Project files are located in projectfiles/uvision
4. Repeat 3 with Python 3
- tools\launch_uvision.bat env37 requirements7.txt
Confirm with the same project locations

0xc0170, 2018-09-24:
I tested compilation for kl26z_bl project. It builds for both python versions I was testing above. There can be a bug in the scripts you are using or version mismatch (mine and you used) ?

0xc0170, 2018-10-12:
I was finally able to reproduce.

`
        uvproj_xml = xmltodict.unparse(uvproj_dic, pretty=True)
`
This line in uvision provides the wrong xml dict, will investigate how it differs between 2.x and 3.x python versions.

closed 2018-10-20 by 0xc0170

0xc0170, 2018-10-20:
v0.9.13 was released

PR #446 (2018-10-20): uvision: fix python3 - remove encode
Should fix #445 

No encoding needed. This turned those to bytes and then unparsed via xmldict to just bytes.