project-generator/project_generator #449: Exporter Request: VS Code build using mbed-cli

40Grit, 2019-06-08:
@trowbidgec @farrenv

We have our own template for this, but having exporter would still be nice if we ever need to validate one of the stock examples with our boards.

0xc0170, 2019-08-14:
Hi @40Grit , do you have the template for sharing? I can have a look. 

As I understand, this request is for adding vs code template that uses mbed-cli? Is this part of Mbed OS exporters ?

40Grit, 2019-08-14:
We are going to stay using @janjongboom's exporter for this so I would start there.

40Grit, 2019-08-14:
*start using

40Grit, 2019-08-14:
https://github.com/janjongboom/mbed-vscode-generator

0xc0170, 2020-01-05:
@40Grit Would you be able to help adding this? I can help but not completely port currently. Otherwise this will be just without progress (I closed some old tool port requests recently).

40Grit, 2020-01-05:
No resources currently to add this into the OS. As Mbed Studio matures we will probably switch over.

Feel free to close unless @janjongboom has some insight proving it a straight forward task.

closed 2020-01-07 by 0xc0170