project-generator/project_generator #451: yaml error

SuWeipeng, 2019-08-03:
errors
---
[image]

how to solve
---
[image]
[image]
[image]

0xc0170, 2019-08-14:
@SuWeipeng Feel free to send PR or I can do it later. 

What version of pyaml you have installed? I'll try to update and reproduce the issue.

SuWeipeng, 2019-08-20:
> @SuWeipeng Feel free to send PR or I can do it later.
> 
> What version of pyaml you have installed? I'll try to update and reproduce the issue.

New installation on windows with newest Python2.7, yaml version greater than 5.1

0xc0170, 2019-08-20:
> @SuWeipeng Feel free to send PR or I can do it later.

If you cand send PR with the fix, I shall find time to update my local packages and get this fix in otherwise

0xc0170, 2019-08-20:
I can reproduce the problem, will send Pr and test

commit eaf72bc8 (2019-08-20): yaml loader: fix #451 - use full loader
commit f43c4a3e (2019-08-20): Merge pull request #452 from project-generator/fix_451

Fix #451
closed 2019-08-20 by 0xc0170