project-generator/project_generator #471: Makefile: Windows paths not escaped

flit, 2020-03-26:
When using make_gcc_arm on Windows, the paths in the generated makefile contain unescaped `\` path separators.

0xc0170, 2020-03-27:
We fix paths using `os.path.normpath`. Locally on windows for frdm example, include paths:

```
progen generate -p gpio_demo_frdmk64f -t make_gcc_arm
progen.generate INFO    Generating make_gcc_arm for gpio_demo_frdmk64f
['..\\..\\gpio_demo_frdmk64f', '..\\..\\cmsis\\MK64F12', '..\\..\\cmsis']
```

But are written unescaped, will look at this over the weekend.

0xc0170, 2020-06-25:
I will close this as I could not reproduce the issue and did not find anything neither that would raise a flag. If it's not the case, lets reopen with an update.

flit, 2020-07-06:
Sorry I didn't notice you closed it… it's a real issue. The problem is that path separators _are_ translated to backslashes on Windows, but the backslashes themselves are not escaped. This causes problems in the makefile output, because make interprets the backslashes as escapes and not path separators. The makefile output needs to double-escape backslashes in paths to work on Windows.

0xc0170, 2020-07-08:
Lets use the same example and compare Make versions and what is there. I am using `https://github.com/project-generator/project_generator_frdm_boards_example` . Using `progen generate -t make_gcc_arm` to get make files.

`C_SRCS :=  ..\..\cmsis\MK64F12\system_MK64F12.c  ..\..\gpio_demo_frdmk64f\main.c ` for example. This is causing a problem on windows?

```
progen generate -t make_gcc_arm
progen.generate INFO    Generating make_gcc_arm for gpio_demo_frdmk64f
progen.generate INFO    Generating make_gcc_arm for gpio_demo_frdmkl25z
progen.generate INFO    Generating make_gcc_arm for gpio_demo_frdmkl46z

project_generator_frdm_boards_example (master -> origin)
cd generated_projects\make_gcc_arm_gpio_demo_frdmk64f\

project_generator_frdm_boards_example\generated_projects\make_gcc_arm_gpio_demo_frdmk64f (master -> origin)
make
Preprocessing ..\..\cmsis\MK64F12\MK64F12.ld
Compiling ..\..\cmsis\MK64F12/system_MK64F12.c
Compiling ..\..\gpio_demo_frdmk64f/main.c
Assembling ..\..\cmsis\MK64F12/startup_MK64F12.S
Linking build/gpio_demo_frdmk64f.elf
   text    data     bss     dec     hex filename
   2108    1096      61    3265     cc1 build/gpio_demo_frdmk64f.elf
   2108    1096      61    3265     cc1 (TOTALS)
Converting build/gpio_demo_frdmk64f.hex
Converting build/gpio_demo_frdmk64f.bin
```

Versions:
progen v0.10 from master
GNU Make 3.81

tested on windows 10, in the cmder.

I just in case tested in the cmd windows and it fails
```
make
process_begin: CreateProcess(NULL, true, ...) failed.
process_begin: CreateProcess(NULL, mkdir build, ...) failed.
make (e=2): The system cannot find the file specified.
make: [build] Error 2 (ignored)
Preprocessing ..\..\cmsis\MK64F12\MK64F12.ld
cc1.exe: fatal error: opening output file build/gpio_demo_frdmk64f.generated.ld: No such file or directory
compilation terminated.
make: *** [build/gpio_demo_frdmk64f.generated.ld] Error 1

```

This needs a fix

flit, 2020-07-08:
Thanks. I'll try it out with the example projects (didn't even realize those existed!). Will be a little bit before I can get to it, though. And I'll have to start up my Windows VM before I can check the version of make.

0xc0170, 2020-07-08:
Ah wait, cmd in windows is having a problem to create build folder. Once I create it manually, I get a binary:

```
Preprocessing ..\..\cmsis\MK64F12\MK64F12.ld
Compiling ..\..\cmsis\MK64F12/system_MK64F12.c
Compiling ..\..\gpio_demo_frdmk64f/main.c
Assembling ..\..\cmsis\MK64F12/startup_MK64F12.S
Linking build/gpio_demo_frdmk64f.elf
   text    data     bss     dec     hex filename
   2108    1096      61    3265     cc1 build/gpio_demo_frdmk64f.elf
   2108    1096      61    3265     cc1 (TOTALS)
Converting build/gpio_demo_frdmk64f.hex
Converting build/gpio_demo_frdmk64f.bin
```

0xc0170, 2020-07-08:
> Thanks. I'll try it out with the example projects (didn't even realize those existed!)

I realized we did not have any showcase for progen so created very simple projects (I ported my baremetal frdm examples).

flit, 2020-07-08:
But it still reports a 'cannot find the file specified' error?

Anyway, it's clear there's some different! I was getting file not found errors because the backslashes weren't double-escaped.

flit, 2020-07-08:
I think you've given me enough so I can try to track it down on my side. There may still be an issue here, or may not. 😄  It would be nice to work with all common make implementations for Windows—I'd be really surprised if I had a weird, unusual version installed.

0xc0170, 2020-07-08:
```
process_begin: CreateProcess(NULL, true, ...) failed.
mkdir build
process_begin: CreateProcess(NULL, mkdir build, ...) failed.
make (e=2): The system cannot find the file specified.
make: [build] Error 2 (ignored)
Preprocessing ..\..\cmsis\MK64F12\MK64F12.ld
arm-none-eabi-cpp -E -x c -P -I.  -I..\..\gpio_demo_frdmk64f  -I..\..\cmsis\MK64F12  -I..\..\cmsis -DNDEBUG  -MMD ..\..\cmsis\MK64F12\MK64F12.ld -o build/gpio_demo_frdmk64f.generated.ld
cc1.exe: fatal error: opening output file build/gpio_demo_frdmk64f.generated.ld: No such file or directory
compilation terminated.
make: *** [build/gpio_demo_frdmk64f.generated.ld] Error 1
```

The first error I believe comes from:
```
PRE_BUILD_SCRIPT := $(shell true)

all: $(ALL_TARGET_OUT_FILES)
	$(at) true
```

The second:
```
$(OUT_DIR):
ifeq ($(OS),Windows_NT)
	$(at)-mkdir $(OUT_DIR)
else
	$(at)$(shell mkdir $(OBJ_FOLDER) 2>/dev/null)
endif
```

And the last one:

```
Converting build/gpio_demo_frdmk64f.hex
Converting build/gpio_demo_frdmk64f.bin
true
process_begin: CreateProcess(NULL, true, ...) failed.
make (e=2): The system cannot find the file specified.
make: *** [all] Error 2
```

Not sure why is this executed after binary creation but again its related to shell true command as we can see from the process creation.

Hope this helps.

0xc0170, 2021-11-04:
Closing as it's unclear if it's still an issue or not. Assuming the latter is the case here.

flit, 2021-11-04:
Oddly enough, I was just having trouble building DAPLink under Windows due to this issue (or at least an issue with Windows paths not being escaped). I tried with both versions of GNU Make for Windows: GNU Make 3.81 for Win32, and GNU Make 4.3 as installed by choco. But previously I was pretty certain I had been able to build DAPLink on Windows using one of those versions of make. No idea…