project-generator/project_generator #477: pre_build_script and post_build_script API

mbrossard, 2020-12-18:
Looking at the `DAPLink` project it seems the different pre-build and post-build scripts have an implied calling convention: they should be called with a build directory named after the current project that needs to be 3 directories down from top-level with a `build` sub-directory with the binaries.

I think the scripts should be given explicitly the different parameters:
- Identifier of the project
- Top-level directory
- Binaries directory
- Verbosity (optional)

I can see two implementation strategies:
- Define variables that can be added to the command line in the YAML configuration
- Add those parameter on the command line to all calls

0xc0170, 2021-01-14:
Can you share references of these in DAPlink, what they do and how ? I do not fully understand the usage of these in the applications.

mathias-arm, 2021-01-14:
Currently DAPLink uses wrapper shell scripts like:
- https://github.com/ARMmbed/DAPLink/blob/master/tools/pre_build_script.sh
- https://github.com/ARMmbed/DAPLink/blob/master/tools/post_build_script.sh

They use wildcard for input and convention (`firmware`) for outputs. This branch uses the name of current working directory to get the project name: https://github.com/mbrossard/DAPLink/blob/feature/compilers/tools/post_build_script_gcc.py