project-generator/project_generator #482: Support for per-file flags

flit, 2021-03-10:
Some projects require per-file compiler flags.

This should probably work by allowing a source file entry to be a usual settings dict.

0xc0170, 2021-03-14:
I recall working on some projects that had specific files settings. I always wanted to avoid this (it is hidden from common settings). 

The support for all tools will take the most effort.

flit, 2021-03-14:
Yeah, it's a pain. Unfortunately it's needed for some things I'm working on in Research. Currently I'm working around it with a new tool-specific `custom_rules` setting added in my fork. It accepts a string that is just copied directly into the generated `Makefile` or `CMakeLists.txt`.

eleenest, 2022-03-10:
Just curious if this feature is still in the works.  

Not urgent, but we did have a case where we wanted to apply flags to only some files (i.e. enable -Werror for files that have been cleaned up, so they stay clean).  

Eric