project-generator/project_generator #486: Release wheels for progen

flit, 2021-03-21:
Need to release wheels as well as sdist for progen. Same for project-generator-definitions.

When pip installs progen now, it will either complain about 

    Using legacy 'setup.py install' for project-generator, since package 'wheel' is not installed.
    Using legacy 'setup.py install' for project-generator-definitions, since package 'wheel' is not installed.

or

    Building wheels for collected packages: project-generator, project-generator-definitions

depending on whether the `wheel` package is installed. (It might not be installed in a fresh virtualenv.)

----

Also need to update the progen Releases in GitHub.

0xc0170, 2021-04-02:
I'll fix the wheels, I have missed pushing in the recent releases

> Also need to update the progen Releases in GitHub.

Start using it to describe the releases ? Or is there anything else we can improve?

0xc0170, 2021-04-05:
I released wheels for 0.9.17, 0.10.0 and 0.11.x releases.

I'll fix progen definitions now as well.

I created also 0.11.x releases on Github.

flit, 2021-04-05:
The comment about GitHub releases was so that, a) people see the latest release version when looking at the project, b) we can record release notes. Looks great, thanks!

closed 2021-04-05 by flit