project-generator/project_generator #488: LD_SCRIPT_IN = []

Andy-1211, 2021-04-07:
Hi.

When I used Makefile generated by "progen generate -f .\projects.yaml -t make_gcc_arm", it tells me the LD_SCRIPT_IN = [].
By the way, the yaml file is geberated by "progen init" form keil project, and I didn't modify anything.
Should I modify LD_SCRIPT_IN, or something else?

Thank you for taking time out of your busy schedule to answer my questions.

0xc0170, 2021-04-26:
if you run init, is there linker script? Can you check yaml files, if there is anywhere? Was it picked up?

0xc0170, 2021-04-26:
It should be created with https://github.com/project-generator/project_generator/blob/master/project_generator/init_yaml.py#L92 (if found).

init functionality is to speed up porting a new application. It is not 100 accurate from the experience. We added it to get the base for the project rather than handcrafting yaml files.