project-generator/project_generator #496: Test suite and `progen list` failing (argument -f/--file: projects.yaml is not a file.)

knowledgejunkie, 2022-02-05:
### Build Test Failures

I'm packaging project_generator 0.11.2 for Debian (testing with Python 3.9/3.10) and the test suite is failing on 3 TestListCommand tests.

These issues seem to be related to the recent changes in 86fcd76cb4fce7fd19881c83f79694bc466ccb7f

```
=========================== short test summary info ============================               
FAILED tests/test_commands/test_list.py::TestListCommand::test_list_projects                   
FAILED tests/test_commands/test_list.py::TestListCommand::test_list_targets                    
FAILED tests/test_commands/test_list.py::TestListCommand::test_list_tools - S...               
================== 3 failed, 71 passed, 13 warnings in 2.09s ===================
```

Example (trimmed) debug output from a test failure:

```
______________________ TestListCommand.test_list_projects ______________________                                                                                                               
                                                                                                                                                                                               
self = ArgumentParser(prog='__main__.py list', usage=None, description=None, formatter_class=<class 'argparse.HelpFormatter'>, conflict_handler='error', add_help=True)                        
action = _StoreAction(option_strings=['-f', '--file'], dest='file', nargs=None, const=None, default='projects.yaml', type=<function argparse_filestring_type at 0x7f9542533f40>, choices=None, 
help='YAML projects file', metavar=None)                                                                                                                                                       
arg_string = 'projects.yaml'                                                                                                                                                                   
                                                                                                                                                                                               
    def _get_value(self, action, arg_string):                                                                                                                                                  
        type_func = self._registry_get('type', action.type, action.type)                                                                                                                       
        if not callable(type_func):                                                                                                                                                            
            msg = _('%r is not callable')                                                                                                                                                      
            raise ArgumentError(action, msg % type_func)                                                                                                                                       
                                                                                                                                                                                               
        # convert the value to the appropriate type                                                                                                                                            
        try:                                                                                                                                                                                   
>           result = type_func(arg_string)                                                                                                                                                     
                                                                                                                                                                                               
/usr/lib/python3.10/argparse.py:2495:                                                                                                                                                          
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _                                                                                                                
                                                                                                                                                                                               
string = 'projects.yaml'

    def argparse_filestring_type(string):
        if not exists(string):
>           raise argparse.ArgumentTypeError("%s is not a file." % string)
E           argparse.ArgumentTypeError: projects.yaml is not a file.

project_generator/commands/__init__.py:20: ArgumentTypeError
```

### Runtime Failures

I tested further and noticed that `progen list tools|targets|prokects` is also failing on 0.11.2, whilst working on 0.11.1.

**WORKING** 0.11.1 installed via pip in a venv:

```
$ progen list tools                    

Progen supports the following tools:

iar_arm
uvision4
uvision5
coide
make_gcc_arm
make_armcc
make_armclang
eclipse_make_gcc_arm
sublime_make_gcc_arm
gdb
arm_none_eabi_gdb
jlink_gdb
cmake_gcc_arm
cmake_armcc
cmake_armclang
visual_studio_gdb
visual_studio_make_gcc_arm
uvision
iar
make_gcc
gcc_arm
armcc
armclang
eclipse
sublime
sublime_text
visual_studio
cmake
gcc_arm
uvision
armcc
iar
armclang
```

**FAILING** 0.11.2 installed via pip in a venv:

```
$ progen list tools
usage: progen list [-h] [-v] [-q] [-f FILE] [-a] [{targets,tools,projects}]
progen list: error: argument -f/--file: projects.yaml is not a file.
```

Given that TravisCI is no longer testing, are these failures repeatable for anyone else?

0xc0170, 2022-02-27:
Thanks for the report. I have to move to Gitlab CI the next week and I'll will address this.

0xc0170, 2022-02-27:
Got it. The commit 

> These issues seem to be related to the recent changes in https://github.com/project-generator/project_generator/commit/86fcd76cb4fce7fd19881c83f79694bc466ccb7f

broke the test. The behavior was not correct in the test itself. I created #499 to address this.

0xc0170, 2022-02-27:
@flit Commit 86fcd76cb4fce7fd19881c83f79694bc466ccb7f adds default to -f. But if we use `--all` argument, we want to check what progen supports, `-f` should not be required in that case. The tests started failing because of this. I would like to verify what is the intention there with default argument.

See PR #499 (it's draft, I'll clean the history tomorrow)

0xc0170, 2022-02-28:
#499 was merged, I am releasing a new version now to address this breakage. Closing as fixed.

0xc0170, 2022-02-28:
@knowledgejunkie appreciate the detailed report , it helped me to fix this quickly

closed 2022-02-28 by 0xc0170

0xc0170, 2022-02-28:
0.11.3 released