project-generator/project_generator #497: Proper usage message output

elfmimi, 2022-02-06:
Executing `progen` without any arguments shows error message:
> Traceback (most recent call last):
  File "/home/username/VENV/bin/progen", line 8, in <module>
    sys.exit(main())
  File "/home/username/VENV/lib/python3.9/site-packages/project_generator/__main__.py", line 70, in main
    return args.func(args)
AttributeError: 'Namespace' object has no attribute 'func'

I suppose something like this is required

```python
    def print_usage(args):
        parser.print_usage()
    parser.set_defaults(func=print_usage)
```

before this line.

https://github.com/project-generator/project_generator/blob/34d9d7b4cabd053404e8560c0479cfc6d597d1ae/project_generator/__main__.py#L51

0xc0170, 2022-02-27:
Thanks for the report, Ill check

0xc0170, 2022-02-27:
@elfmimi I am fixing some other errors today. Feel free to send a pull request, I'll review.

0xc0170, 2022-02-28:
Fixed via #501

closed 2022-02-28 by 0xc0170